bifravst/firmware #1: GPS fix not getting acquired

coderbyheart, 2020-02-05:
The latest relase v0.65.0 is no longer acquiring GPS fixes although satellites are available. It very quickly terminates the GPS:

```
*** Booting Zephyr OS version 2.1.0  ***
[00:00:00.201,049] <dbg> nrf9160_gps.init: MAGPIO set: AT%XMAGPIO=1,0,0,1,1,1574,1577
[00:00:00.216,125] <dbg> nrf9160_gps.init: COEX0 set: AT%XCOEX0=1,1,1570,1580
[00:00:00.223,937] <inf> cat_tracker: The cat tracker has started
[00:00:00.230,651] <inf> cat_tracker: Version: v0.65.0
[00:00:00.243,225] <inf> dfu_target: Device get binding device
[00:00:00.256,286] <inf> gps_control: GPS initialized
[00:00:00.261,932] <inf> cat_tracker: Connecting to LTE network. 
[00:00:00.268,646] <inf> cat_tracker: This may take several minutes.
[00:00:00.289,367] <inf> lte_lc: Using legacy LTE PCO mode...
[00:00:00.295,745] <dbg> lte_lc.w_lte_lc_connect: Network mode: AT%XSYSTEMMODE=0,1,1,0
[00:00:12.461,517] <inf> cat_tracker: Incoming RSRP status message, RSRP value is 77
[00:00:12.470,428] <dbg> lte_lc.parse_nw_reg_status: Network registration status: 2
[00:00:17.349,060] <dbg> lte_lc.parse_nw_reg_status: Network registration status: 5
[00:00:17.357,574] <inf> cat_tracker: Connected to LTE network
[00:00:17.364,074] <dbg> nrf9160_timestamp.nrf9160_time_handler: New date time update in: 3600 seconds
[00:00:17.374,053] <dbg> nrf9160_timestamp.update_new_date_time: Updating date time UTC...
[00:00:17.382,995] <dbg> nrf9160_timestamp.check_current_time: Date time never set
[00:00:17.391,174] <dbg> nrf9160_timestamp.update_new_date_time: Error getting current time
[00:00:17.400,177] <dbg> nrf9160_timestamp.update_new_date_time: Fallback on cellular network time
[00:00:17.411,071] <dbg> nrf9160_timestamp.get_time_cellular_network: Response from modem: +CCLK: "20/02/03,11:24:01+04
[00:00:17.422,576] <dbg> nrf9160_timestamp.update_new_date_time: Time from cellular network successfully obtained
[00:00:18.054,718] <dbg> bifravst_cloud.broker_init: IPv4 Address found 18.196.54.227
[00:00:24.590,942] <inf> aws_jobs: Subscribe: $aws/things/352656100378903/jobs/notify-next
[00:00:24.601,013] <inf> aws_jobs: Subscribe: $aws/things/352656100378903/jobs/$next/get/#
[00:00:24.610,839] <dbg> bifravst_cloud.mqtt_evt_handler: MQTT client connected!
[00:00:24.618,835] <dbg> bifravst_cloud.topic_subscribe: Subscribing to: $aws/things/352656100378903/shadow/get/rejected
[00:00:24.630,401] <dbg> bifravst_cloud.topic_subscribe: Subscribing to: $aws/things/352656100378903/shadow/get/accepted/desired/cfg
[00:00:24.643,066] <dbg> bifravst_cloud.topic_subscribe: Subscribing to: $aws/things/352656100378903/shadow/update/delta
[00:00:24.655,578] <inf> cat_tracker: CLOUD_EVT_CONNECTED
[00:00:24.661,621] <dbg> bifravst_cloud.data_publish: Publishing to topic: $aws/things/352656100378903/shadow/get
[00:00:24.870,880] <inf> aws_fota: subscribed to notify-next topic
[00:00:24.877,746] <inf> aws_jobs: Publish topic: $aws/things/352656100378903/jobs/$next/get
[00:00:24.886,810] <inf> aws_jobs: Publish payload {"clientToken": ""}
[00:00:25.066,101] <dbg> bifravst_cloud.mqtt_evt_handler: MQTT_EVT_SUBACK: id = 2114 result = 0
[00:00:25.118,804] <dbg> bifravst_cloud.mqtt_evt_handler: MQTT_EVT_SUBACK: id = 1234 result = 0
[00:00:25.479,339] <dbg> aws_fota.aws_fota_on_publish_evt: Received topic: $aws/things/352656100378903/shadow/get/accepted/desired/cfg
[00:00:25.492,156] <inf> aws_fota: received an unhandled MQTT publish event on topic: $aws/things/352656100378903/shadow/get/accepted/desired/cfg
[00:00:25.505,920] <dbg> bifravst_cloud.mqtt_evt_handler: MQTT_EVT_PUBLISH: id = 15764 len = 74 
[00:00:25.515,441] <inf> cat_tracker: CLOUD_EVT_DATA_RECEIVED
[00:00:25.523,315] <inf> cloud_codec: Decoded message: {
                                                                "cfg":  {
                                                                                        "act":  true,
                                                                                                                "mvres":        60,
                                                                                                                                                "mvt":  3590,
                                                                                                                                                                 "acct": 10,
                                                                                                                                                                              "actwt": 60,
                                                                                                                                                                                            "gpst": 600
                                                                                                                                                                                                        }
                                                                                                                                                                                                         }
[00:00:25.538,421] <inf> cloud_codec: SETTING GPST TO: 600
[00:00:25.544,494] <inf> cloud_codec: SETTING ACTIVE TO: 1
[00:00:25.550,567] <inf> cloud_codec: SETTING ACTIVE WAIT TO: 60
[00:00:25.557,159] <inf> cloud_codec: SETTING PASSIVE_WAIT TO: 60
[00:00:25.563,842] <inf> cloud_codec: SETTING MOVEMENT TIMEOUT TO: 3590
[00:00:25.571,075] <inf> cloud_codec: SETTING ACCEL THRESHOLD TIMEOUT TO: 10
[00:00:25.594,055] <dbg> bifravst_cloud.mqtt_evt_handler: MQTT_EVT_PUBACK: id = 29293 result = 0
[00:00:25.679,077] <dbg> aws_fota.aws_fota_on_publish_evt: Received topic: $aws/things/352656100378903/jobs/$next/get/acceptedsired/cfg
[00:00:25.691,955] <inf> aws_fota: Checking for an available job
[00:00:25.698,669] <dbg> aws_fota.get_job_execution: Got only one field: {"clientToken":"
[00:00:25.707,489] <inf> aws_fota: No queued jobs for this device
[00:00:29.674,163] <inf> cloud_codec: Encoded message: {
                                                                "state":        {
                                                                                                "reported":     {
                                                                                                                                        "cfg":  {
                                                                                                                                                                  "gpst": 600,
                                                                                                                                                                                  "act": true,
[... 181 lines cut ...]
[00:11:40.089,202] <dbg> nrf9160_gps.print_satellite_stats: Tracking SV 20: not used,     healthy
[00:11:40.098,724] <dbg> nrf9160_gps.print_satellite_stats: Tracking SV  8: not used,     healthy
[00:11:40.108,245] <dbg> nrf9160_gps.print_satellite_stats: Tracking: 3 Using: 0 Unhealthy: 0
[00:11:40.117,431] <dbg> nrf9160_gps.print_satellite_stats: Seconds since last fix 700
[00:11:41.079,833] <dbg> nrf9160_gps.gps_thread: Waiting for time window to operate
[00:13:48.822,296] <dbg> nrf9160_gps.gps_thread: Invalid PVT frame, discarding
[00:13:49.862,304] <dbg> nrf9160_gps.gps_thread: Invalid PVT frame, discarding
[00:13:50.862,518] <dbg> nrf9160_gps.gps_thread: Invalid PVT frame, discarding
[00:13:51.658,966] <inf> cat_tracker: Incoming RSRP status message, RSRP value is 77
[00:13:51.862,640] <dbg> nrf9160_gps.gps_thread: Invalid PVT frame, discarding
[00:13:52.863,098] <dbg> nrf9160_gps.gps_thread: Invalid PVT frame, discarding
[00:13:53.863,098] <dbg> nrf9160_gps.gps_thread: Invalid PVT frame, discarding
[00:13:54.863,677] <dbg> nrf9160_gps.gps_thread: Invalid PVT frame, discarding
[00:13:55.863,708] <dbg> nrf9160_gps.gps_thread: Invalid PVT frame, discarding
[00:13:56.863,769] <dbg> nrf9160_gps.gps_thread: Invalid PVT frame, discarding
[00:13:57.864,013] <dbg> nrf9160_gps.gps_thread: Invalid PVT frame, discarding
[00:13:58.864,135] <dbg> nrf9160_gps.gps_thread: Invalid PVT frame, discarding
[00:13:59.864,807] <dbg> nrf9160_gps.gps_thread: Invalid PVT frame, discarding
[00:14:00.864,837] <dbg> nrf9160_gps.gps_thread: Invalid PVT frame, discarding
[00:14:01.865,112] <dbg> nrf9160_gps.gps_thread: Invalid PVT frame, discarding
[00:14:02.865,814] <dbg> nrf9160_gps.gps_thread: Invalid PVT frame, discarding
[00:14:03.115,081] <dbg> nrf9160_gps.print_satellite_stats: Tracking: 0 Using: 0 Unhealthy: 0
[00:14:03.124,298] <dbg> nrf9160_gps.print_satellite_stats: Seconds since last fix 843
[00:14:04.120,697] <dbg> nrf9160_gps.gps_thread: Waiting for time window to operate
```

simensrostad, 2020-02-06:
What do you mean by quickly terminates the GPS? From the looks of it the device simply does not get a fix which is a common occurrence.

coderbyheart, 2020-02-10:
Right, looking at the timing now, I see that it takes quite a while to search for GPS, it does not log much during that...

closed 2020-02-10 by coderbyheart