bifravst/firmware #5: Push button: send message to help identify device

coderbyheart, 2020-02-05:
When the Button 1 of the device is released it should publish a message with a timestamp to a custom topic.

```json
{
    "btn": {
      "v": 1,
      "ts": 1563968747123
  }
}
```

`v` is the number of the button
`ts` is the timestamp of the release

*Note:* we deliberately do not use the shadow here to

1. demonstrate how to use custom topics and
1. because the button itself is a UI element and not a sensor.

simensrostad, 2020-03-14:
Lets agree on a custom topic here if that has not been defined yet. @coderbyheart.

simensrostad, 2020-03-14:
Maybe thingname/notify.

coderbyheart, 2020-03-16:
I would suggest something more general: `<clientId>/messages`.

simensrostad, 2020-03-16:
Ok, that works. It seems reasonable given the fact that the button can be used to support a range of different functionalities.

closed 2020-03-17 by simensrostad

coderbyheart, 2020-03-18:
Note, the button is not reported on release, but on push, but this is a minor issue.

coderbyheart, 2020-03-18:
On the 91DK only button 1 works.

commit 643135f9 (2020-03-17): feat: added button functionality

Button one will publish a message to the cloud
signifying that it has been pressed.

Button two gives the accelerometer semaphore essentially
simulating movement in the case of nRF9160 DK.
This is becuase the nRF9160 DK does not have a accelerometer
by default. Closes #5 and closes #6