bifravst/firmware #21: Unreliable connection to AWS + freezing issue

levic92, 2020-03-03:
I have setup the project and compiled the firmware to connect to my AWS account. There are 2 issues I am having

1. The Thingy:91 will publish a few updates to AWS and then the updates stop coming in. Even though I'm not seeing any data coming to AWS I can see in the LTE Monitor log that the Thingy:91 is still acquiring new GPS positions and the log even says it is publishing to AWS.

2. Then after a period of time (usually less than an hour) the Thingy:91 just freezes and no more logs are generated, and most of the time the LED is stuck on.

**Here is the log from the first test:**
[Log_1.log](https://github.com/bifravst/firmware/files/4284034/Log_1.log)
- no data in AWS after 2020-03-03T18:42:07.1Z
- first 2 locations sent but the rest failed
- last line of the log is when it froze

**LED stuck on once logs stop**
[image]

**Here is the log from the second test:**
[Log_2.log](https://github.com/bifravst/firmware/files/4284035/Log_2.log)
- no data in AWS after 2020-03-03T20:24:13.9Z
- first 11 locations sent but the rest failed
- last line of the log is when it froze

**LED stuck on once logs stop**
[image]

**Images from AWS IOT showing no data coming in after** 
[image]

**Activity** 
[image]

coderbyheart, 2020-03-04:
Hei @levic92, thanks for reporting this here. It's hard to pinpoint the reason for this, but it's most likely not an issue with the cat tracker firmware, but an issue with a combination of network operator, network mode, and SIM card provider. It could also be NAT timeout causing ACKs to never show up and buffers in the mode filling up and eventually blocking the system. The best place (and the most competent people) to help with these issues is [{DevZone](https://devzone.nordicsemi.com/), please open a case there.

Modem traces will be very helpful to further debug this. The new [connectivity bridge](https://github.com/NordicPlayground/fw-nrfconnect-nrf/pull/1960) can be used to capture traces from Thingy:91.

levic92, 2020-03-04:
How would you explain the device freezing and no longer producing logs?

coderbyheart, 2020-03-24:
Closing because of inactivity.

closed 2020-03-24 by coderbyheart