bifravst/firmware #23: Accelerometor readings are wrong, also wrong timestamp

coderbyheart, 2020-03-06:
These are 4 different readings, note how they alternated between two states, also the timestamp is wrong: `3167027337385` seems like a timestamp is added to another timestamp.

```
"acc":  {
        "v":    [-2.667408, -0.48052499999999998, -10.051816000000001],
        "ts":   3167027337385
}
"acc":  {
        "v":    [-2.667408, -0.48052499999999998, -10.051816000000001],
        "ts":   1583528740126
}
"acc":  {
        "v":    [-1.098344, 1.7161629999999999, -9.5712899999999994],
        "ts":   3167026855055
}
"acc":  {
        "v":    [-1.098344, 1.7161629999999999, -9.5712899999999994],
        "ts":   1583528257796
}
[08:43:02.911,712] <inf> cat_tracker: Going to sleep for: 60 seconds
Encoded message: {
        "state":        {
                "reported":     {
                        "bat":  {
                                "v":    4516,
                                "ts":   1583529980271
                        },
                        "acc":  {
                                "v":    [-1.5298370000000001, 0, -10.208722],
                                "ts":   3167028321119
                        }
                }
        }
}

```

coderbyheart, 2020-03-24:
This is still broken in v1.6.2:

```
{"reported":{"bat":{"v":3961,"ts":1584924362613},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":63391498760355}},"timestamp":"2020-03-23 00:46:06.2","deviceId":"352656100248049"}
{"reported":{"bat":{"v":3961,"ts":1584927962645},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":64976286227760}},"timestamp":"2020-03-23 01:46:14.6","deviceId":"352656100248049"}
{"reported":{"bat":{"v":3961,"ts":1584931561675},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":66561073694147}},"timestamp":"2020-03-23 02:46:04.9","deviceId":"352656100248049"}
{"reported":{"bat":{"v":3957,"ts":1584935161712},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":68145861160534}},"timestamp":"2020-03-23 03:46:09.4","deviceId":"352656100248049"}
{"reported":{"bat":{"v":3957,"ts":1584938761731},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":69730648626903}},"timestamp":"2020-03-23 04:46:09.6","deviceId":"352656100248049"}
{"reported":{"bat":{"v":3957,"ts":1584942361757},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":71315436093272}},"timestamp":"2020-03-23 05:46:08.1","deviceId":"352656100248049"}
{"reported":{"bat":{"v":3953,"ts":1584945961776},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":72900223559623}},"timestamp":"2020-03-23 06:46:07.4","deviceId":"352656100248049"}
{"reported":{"bat":{"v":3953,"ts":1584949561813},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":74485011025974}},"timestamp":"2020-03-23 07:46:06.8","deviceId":"352656100248049"}
{"reported":{"bat":{"v":3953,"ts":1584953161843},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":76069798492307}},"timestamp":"2020-03-23 08:46:06.8","deviceId":"352656100248049"}
{"reported":{"bat":{"v":3945,"ts":1584956761880},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":77654585958640}},"timestamp":"2020-03-23 09:46:05.5","deviceId":"352656100248049"}
{"reported":{"bat":{"v":3949,"ts":1584960361900},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":79239373424955}},"timestamp":"2020-03-23 10:47:52.1","deviceId":"352656100248049"}
{"reported":{"bat":{"v":3949,"ts":1584963961937},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":80824160891270}},"timestamp":"2020-03-23 11:46:04.2","deviceId":"352656100248049"}
{"reported":{"bat":{"v":3945,"ts":1584967560956},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":82408948356567}},"timestamp":"2020-03-23 12:46:08.7","deviceId":"352656100248049"}
{"reported":{"bat":{"v":3941,"ts":1584971160993},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":83993735821864}},"timestamp":"2020-03-23 13:46:08.1","deviceId":"352656100248049"}
{"reported":{"bat":{"v":3941,"ts":1584974761000},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":85578523287143}},"timestamp":"2020-03-23 14:46:06.4","deviceId":"352656100248049"}
{"reported":{"bat":{"v":3941,"ts":1584978361049},"acc":{"v":[1.6279029999999999,-5.5995970000000002,-11.38552],"ts":87163310752422}},"timestamp":"2020-03-23 15:46:09.0","deviceId":"352656100248049"}
{"reported":{"dev":{"v":{"band":20,"nw":"NB-IoT GPS","iccid":"89450421180216254864","modV":"mfw_nrf9160_1.1.1","brdV":"nrf9160_pca20035","appV":"1.6.2"},"ts":1584980907091},"roam":{"v":{"rsrp":55,"area":3305,"mccmnc":24202,"cell":34247442,"ip":"10.81.185.242"},"ts":1584980907091}},"timestamp":"2020-03-23 16:28:28.7","deviceId":"352656100248049"}
```

coderbyheart, 2020-03-24:
It looks like the readings are not reset, but kept in memory:

```
[00:11:36.311,492] <inf> cat_tracker: Going to sleep for: 60 seconds
Encoded message: {
        "state":        {
                "reported":     {
                        "bat":  {
                                "v":    4348,
                                "ts":   1585062863097
                        },
                        "acc":  {
                                "v":    [-0.019612999999999998, 0.078452999999999995, -9.8949090000000002],
                                "ts":   1585062837550
                        }
                }
        }
}
[00:11:36.437,011] <dbg> aws_iot.aws_iot_send: Publishing to topic: $aws/things/352656100248049/shadow/update

<later>

[00:22:36.370,361] <inf> cat_tracker: Going to sleep for: 60 seconds
Encoded message: {
        "state":        {
                "reported":     {
                        "bat":  {
                                "v":    4348,
                                "ts":   1585063523156
                        },
                        "acc":  {
                                "v":    [-0.019612999999999998, 0.078452999999999995, -9.8949090000000002],
                                "ts":   3170125004235
                        }
                }
        }
}
[00:22:36.495,880] <dbg> aws_iot.aws_iot_send: Publishing to topic: $aws/things/352656100248049/shadow/update
```

Exact same readings reported, but timestamps are added.

simensrostad, 2020-03-24:
Does the problem still persist given the latest commit. 615cd854e9830fc17c99533ec14c36268a061e7f ?

simensrostad, 2020-03-24:
``[2.5987619999999998, 1.1081509999999999, -14.906108]``
``[0.941438, 0.558979, -14.474615]``
``[1.72597, 1.5984829999999999, -11.042287]``

These are three different accelerometer readings transmitted to cloud from my device.

commit e270232a (2020-03-10): fix: removed static keyword to prevent persisting
accelerometer data values. Closes #23
commit 3b4fa169 (2020-03-10): fix: Removed keyword

Removed static keyword to prevent persisting
accelerometer data values. Closes #23
commit babce3a3 (2020-03-14): fix: Removed keyword

Removed static keyword to prevent persisting
accelerometer data values. Closes #23
closed 2020-03-25 by simensrostad