bifravst/firmware #26: running github acttions on cloned repo

cmedianu, 2020-03-12:
Hi Markus,

Do you have any idea why semantic-release won't release on a cloned repo?

https://github.com/cmedianu/bifravst_firmware/runs/502525322
It says:
[semantic-release] › ℹ  The local branch saga is behind the remote one, therefore a new version won't be published.

but that's not true, the branch was just synced.. Any idea how I can force it to build a release regardless?

Also, what does bifravst stand for? :)

Thanks!

coderbyheart, 2020-03-12:
Hei @cmedianu, thanks for reaching out.

By default `semantic-release` will get the `reposityUrl` from the `package.json` which in your case [still points to `https://github.com/bifravst/firmware.git`](https://github.com/cmedianu/bifravst_firmware/blob/80f1267807cca7e5f538635766c17601ec948219/package.json#L7). 

So either, update that to your URL, or [provide it as an argument to the `semantic-release` CLI](https://semantic-release.gitbook.io/semantic-release/usage/configuration#repositoryurl).

coderbyheart, 2020-03-12:
> Also, what does bifravst stand for? :)

*Bifravst* is from the Nordic mythology and refers to the bridge built by the gods between earth and heaven: https://www.lyngen.com/2020/03/the-vikings-and-the-northern-lights-bridge/

closed 2020-03-12 by coderbyheart