bifravst/firmware #30: cloud_connect() fails after going offline and then online again

comemaster, 2020-03-19:
Hello,

we have recently moved to the new Docker based build system and are thus trying to update our code to use a more recent version of nrf-connect. Since we need the `AWS_FOTA_START` event as well as the `date_time.h` lib, we set the same revision in `west.yml` as in the most recent bifravst commit:
```yml
revision: 50e8f1bfe4602932065542721fae2be43e14b623
```

Using newer revisions of nrf-connect also does not fix our problem.

The problem is in the `cloud_poll(void)` function, where execution gets stuck at `err = cloud_connect(cloud_backend);`
This only happens after we disconnect from the network and reconnect again, using `lte_lc_offline();` and `lte_lc_init_and_connect();` respectively.
I modified `cloud_poll` in such a way, that it only breaks the while loop if a cloud disconnect (`POLLNVAL, POLLHUP, POLLERR`) happens. Thus, the thread continues, disconnects from cloud and then waits for the semaphore (`cloud_conn_sem`) again.

Disregarding how cloud_poll exactly functions, the following minimal example demonstrates what happens:

```c
void main(void)
{
    err = modem_data_init();
    if (err)
    {
        LOG_ERR("modem_data_init, error: %d\n", err);
    }

    err = cloud_setup();
    if (err)
    {
        LOG_INF("cloud_setup, error %d", err);
        error_handler(err);
    }

    //TEST
    printk("TEST\n");
    lte_lc_init_and_connect();
    printk("connected to lte\n");

    cloud_connect(cloud_backend);
    printk("connected to cloud\n");
    cloud_disconnect(cloud_backend);
    printk("disconnected from cloud\n");
    cloud_connect(cloud_backend);
    printk("connected to cloud\n");
    cloud_disconnect(cloud_backend);
    printk("disconnected from cloud\n");

    // go offline
    lte_lc_offline();
    printk("went offline\n");

    // try same thig again
    lte_lc_init_and_connect();
    printk("connected to lte\n");

    cloud_connect(cloud_backend);  // code is stuck here

    // none of this happens:
    printk("connected to cloud\n");
    cloud_disconnect(cloud_backend);
    printk("disconnected from cloud\n");
    cloud_connect(cloud_backend);
    printk("connected to cloud\n");
    cloud_disconnect(cloud_backend);
    printk("disconnected from cloud\n");

    printk("TEST END\n");
}
```

Running this example gives the following output:
```
*** Booting Zephyr OS build v2.1.99-ncs1-8-g367eec2dd60d  ***                                                                                 
[00:00:00.193,634] <dbg> nrf9160_gps.setup: MAGPIO set: AT%XMAGPIO=1,0,0,1,1,1574,1577                                                        
[00:00:00.202,453] <dbg> nrf9160_gps.setup: COEX0 set: AT%XCOEX0=1,1,1570,1580                                                                
[00:00:00.217,468] <inf> main: Device IMEI: 352656101002049                                                                                   
TEST                                                                                                                                          
[00:00:00.233,551] <inf> lte_lc: Using legacy LTE PCO mode...                                                                                 
[00:00:02.652,679] <dbg> main.modem_rsrp_handler: Incoming RSRP status message, RSRP value is 58                                              
                                                                                                                                              
connected to lte                                                                                                                              
[00:00:10.934,020] <dbg> aws_iot.broker_init: IPv4 Address found 18.194.141.235                                                               
connected to cloud                                                                                                                            
disconnected from cloud                                                                                                                       
[00:00:22.383,697] <dbg> aws_iot.broker_init: IPv4 Address found 18.194.141.235                                                               
connected to cloud                                                                                                                            
disconnected from cloud                                                                                                                       
went offline                                                                                                                                  
[00:00:35.753,265] <inf> lte_lc: Using legacy LTE PCO mode...                                                                                 
[00:00:38.493,560] <dbg> main.modem_rsrp_handler: Incoming RSRP status message, RSRP value is 58                                              
                                                                                                                                              
connected to lte                                                                                                                              
[00:00:39.487,701] <dbg> aws_iot.broker_init: IPv4 Address found 18.194.141.235   
```
as you can see, execution is stuck in the first cloud_connect() after going offline and back online. 
It seems to get stuck part way through, since a dbg print happens first. If I am correct, this happens [here](https://github.com/NordicPlayground/fw-nrfconnect-nrf/blob/5430013ae375547549e3cb1a1e0900dd6e302875/subsys/net/lib/aws_iot/src/aws_iot.c#L549). Examining [aws_iot_connect](https://github.com/NordicPlayground/fw-nrfconnect-nrf/blob/5430013ae375547549e3cb1a1e0900dd6e302875/subsys/net/lib/aws_iot/src/aws_iot.c#L693) further, maybe it gets stuck in the mqtt connect phase?

I have tried different combination of not calling `cloud_disconnect(cloud_backend)` or calling it, in different orders and such, but nothing seems to change this weird behavior.

Am I using the cloud backend wrong or is this an error in the aws_iot lib?
Any help would be appreciated.

simensrostad, 2020-03-19:
Does the ``cloud_connect()`` function return any error code before the code gets stuck? You could match the returned value with the switch case present in the ``aws_iot``library, line 808. You could also print the socket number in between connect and disconnect to make sure the socket is released when disconnecting and that you are polling on the correct tls socket. ``cloud_backend->config.socket``

comemaster, 2020-03-20:
As stated, execution gets stuck inside `cloud_connect()`. The last thing printed is the `IPv4 Addres found` message.

I added socket prints and also error checking.
Code:
```c
void lte_c()
{
    int err = lte_lc_init_and_connect();
    if (err)
    {
        printk("lte_lc_init_and_connect error: %d\n", err);
    }
    else
    {
        printk("connected to lte\n");
    }
    printk("Socket: %d\n", cloud_backend->config->socket);
}

void lte_dc()
{
    int err = lte_lc_offline();
    if (err)
    {
        printk("lte_lc_offline error: %d\n", err);
    }
    else
    {
        printk("disconnected from lte\n");
    }
    printk("Socket: %d\n", cloud_backend->config->socket);
}

void cloud_c()
{
    int err = cloud_connect(cloud_backend);
    if (err)
    {
        printk("cloud_connect error: %d\n", err);
    }
    else
    {
        printk("connected to cloud\n");
    }
    printk("Socket: %d\n", cloud_backend->config->socket);
}

void cloud_dc()
{
    int err = cloud_disconnect(cloud_backend);
    if (err)
    {
        printk("cloud_disconnect error: %d\n", err);
    }
    else
    {
        printk("disconnected from cloud\n");
    }
    printk("Socket: %d\n", cloud_backend->config->socket);
}

void main(void)
{
    int err;
    err = modem_data_init();
    if (err)
    {
        LOG_ERR("modem_data_init, error: %d\n", err);
    }

[... 82 lines cut ...]
connected to cloud                                                                                                                               
Socket: 4                                                                                                                                        
cloud_disconnect error: -128                                                                                                                     
Socket: 4                                                                                                                                        
disconnected from lte                                                                                                                            
Socket: 4                                                                                                                                        
[00:00:53.957,733] <inf> lte_lc: Using legacy LTE PCO mode...                                                                                    
[00:00:56.529,296] <dbg> main.modem_rsrp_handler: Incoming RSRP status message, RSRP value is 59                                                 
                                                                                                                                                 
connected to lte                                                                                                                                 
Socket: 4                                                                                                                                        
[00:00:57.420,410] <dbg> bifravst_cloud.broker_init: IPv4 Address found 52.58.51.24                                                              
[00:01:05.209,228] <err> bifravst_cloud: mqtt_connect, error: -22                                                                                
cloud_connect error: -22                                                                                                                         
Socket: 4                                                                                                                                        
cloud_disconnect error: -128                                                                                                                     
Socket: 4                                                                                                                                        
[00:01:05.223,358] <dbg> bifravst_cloud.broker_init: IPv4 Address found 52.58.51.24                                                              
[00:01:14.033,355] <err> bifravst_cloud: mqtt_connect, error: -22                                                                                
cloud_connect error: -22                                                                                                                         
Socket: 4                                                                                                                                        
cloud_disconnect error: -128                                                                                                                     
Socket: 4                                                                                                                                        
TEST END 
```

coderbyheart, 2020-03-20:
I have confirmed this internally with my colleagues. `mqtt_client_tls_connect` hangs when it does the call to `connect()` on the socket.
We will look into fixing this as soon as possible.

coderbyheart, 2020-03-24:
It looks like the [missing support for TLS caching](https://github.com/NordicPlayground/nrfxlib/pull/151), which is enabled by default, is causing this.

It _can_ be disabled in code, within the MQTT library, but that's messy, so I recommend to wait for the incoming fix: https://github.com/NordicPlayground/fw-nrfconnect-nrf/pull/2015

Note that the issue with unclosed socket still exists: The disconnect() will always fail if the app never actually reads out data from the MQTT socket. It will then never see the `CONNACK` from the server, and it can therefore not disconnect. That leaves the socket open. The workaround here is to close the socket manually: `close(cloud_backend->config->socket)`.

comemaster, 2020-03-24:
Thank you for looking into this.
We have since reverted to our previous nrf-connect version (based on v1.1 with some changes), and are using our fork+docker build system.
We will probably wait for a new release (v1.3) before trying to upgrade again.

Thanks for the tip on closing the socket, I have added that to our code, just in case. This way, the socket number is always the same, even if cloud_disconnect() fails.

coderbyheart, 2020-04-01:
This should be resolved now.

closed 2020-04-01 by coderbyheart