bifravst/firmware #31: Button press not transmitted on LTE-m

coderbyheart, 2020-03-20:
Despite the device claiming to have sent the message, it never arrives at the broker:

```
[00:26:28.079,315] <dbg> lte_lc.parse_nw_reg_status: Network registration status: 5
[00:26:28.087,585] <inf> cat_tracker: Checking LTE connection...
[00:26:28.094,146] <inf> cat_tracker: LTE link maintained
Encoded message: {
        "btn":  {
                "v":    1,
                "ts":   1584697409409
        }
}
[00:26:28.107,452] <dbg> aws_iot.aws_iot_send: Publishing to topic: 352656100378903/messages
```

coderbyheart, 2020-03-20:
It works with the same version for NB-IoT

simensrostad, 2020-03-20:
I am not able to reproduce. I get it to work in both LTE modes.

coderbyheart, 2020-03-20:
On a DK?

simensrostad, 2020-03-20:
> On a DK?

Yes

simensrostad, 2020-03-20:
``{
  "format": "json",
  "payload": {
    "btn": {
      "v": 1,
      "ts": 1584710132940
    }
  },
  "qos": 0,
  "timestamp": 1584710133607,
  "topic": "352656100382111/messages"
}``

coderbyheart, 2020-03-20:
It works right after boot, it seems to stop working after a while.

simensrostad, 2020-03-20:
I have experienced a prolonged delay after pushing the button and receiving the message in AWS. But that is never more than 5 seconds. I know from previous testing that this delay can be longer.

simensrostad, 2020-03-20:
But I am guessing that is not the case if the message is not received at all.

coderbyheart, 2020-03-25:
I have double checked it now, and I am directly subscribed on the IoT console, the Thingy with NB-IoT (Telia Trio) sends it, the DK with LTE-m (iBasis) does not.

coderbyheart, 2020-12-09:
Tested again with 3.2.2, iBasis SIM and LTE and it works.

closed 2020-12-09 by coderbyheart