bifravst/firmware #35: Connection times out on broker, without device noticing (LTE-m, iBasis)

coderbyheart, 2020-03-25:
[image]

```
[00:04:18.697,723] <inf> cat_tracker: Incoming RSRP status message, RSRP value is 51
[00:27:03.963,989] <inf> cat_tracker: Incoming RSRP status message, RSRP value is 51
[00:27:08.444,244] <inf> cat_tracker: Incoming RSRP status message, RSRP value is 51
[00:29:50.734,558] <inf> cat_tracker: Incoming RSRP status message, RSRP value is 51
[00:30:04.814,758] <inf> cat_tracker: Incoming RSRP status message, RSRP value is 51
[00:40:28.932,128] <dbg> lte_lc.parse_nw_reg_status: Network registration status: 5
[00:40:28.940,429] <inf> cat_tracker: Checking LTE connection...
[00:40:28.947,021] <inf> cat_tracker: LTE link maintained
Encoded message: {
        "btn":  {
                "v":    1,
                "ts":   1585150428875
        }
}
[00:40:28.960,266] <dbg> aws_iot.aws_iot_send: Publishing to topic: 352656100378903/messages
```

After the device reaches the passive timeout, it will reestablish the connection.

coderbyheart, 2020-03-25:
[image]

It reconnects, and now buttons are registered again (#31).

simensrostad, 2020-09-16:
Have not seen this in a while... Lets close it and reopen if this problem persist @coderbyheart ?

closed 2020-09-16 by coderbyheart