bifravst/firmware #40: allow custom initialization to be run at startup

cmedianu, 2020-04-01:
my provider requires me to run  "AT+CGDCONT=0,\"IP\",\"the.apn.name\"" to set the APN on startup. It would be nice to have a configuration option to run some AT commands on startup.

coderbyheart, 2020-04-01:
Agreed, I think it makes sense to have this dynamically configured through the cloud, because these AT commands might even depend on the location of the device (if we have devices roaming in different countries).

cmedianu, 2020-04-01:
That's even better! But it initially has to be compiled in so it can get to the cloud in the first place.

coderbyheart, 2020-04-01:
Agreed, we need both:

1. a compile time firmware configuration setting that accepts a list of AT commands to be run at startup
2. a run time routine that applies a list of AT commands present in the desired device state (after boot and whenever they change)

coderbyheart, 2020-04-01:
> 1. a compile time firmware configuration setting that accepts a list of AT commands to be run at startup

Doing this compile time is supported by adding these configs:

```
CONFIG_LTE_PDP_CMD=y
CONFIG_LTE_PDP_CONTEXT="0,\"IP\",\"the.apn.name\""
```

coderbyheart, 2021-02-04:
I think dynamically modem configuration needs to be handled by the device, based on e.g. network parameters.

closed 2021-02-04 by coderbyheart