bifravst/firmware #46: Firmware does not respond to higher publication intervals

simensrostad, 2020-05-17:
Possible overflow. Changing types of global config structure might help.

coderbyheart, 2020-05-18:
Which intervals are you talking about?

simensrostad, 2020-05-18:
> Which intervals are you talking about?

I did some testing yesterday where I changed the active time to 86400 seconds. The device did report the change back but continued to publish data every hour. (previous setting).

This was something I observed "on the fly". It would be nice to have some confirmation on this issue.

coderbyheart, 2020-05-18:
Yes, can confirm. I set motion timeout to 86400, but it started after an hour again:

```
[2020-05-18 23:51:56] [01:00:00.221,527] <dbg> date_time.new_date_time_get: Time successfully obtained
[2020-05-18 23:51:56] [01:00:00.313,568] <inf> cat_tracker: Movement timeout triggered
[2020-05-18 23:51:56] [01:00:00.320,220] <inf> cat_tracker: The cat is moving!
[2020-05-18 23:51:56] [01:00:00.320,312] <inf> cat_tracker: Movement timeout triggered
```

Then it crashed (probably unrelated):

```
[2020-05-18 23:51:56] ASSERTION FAIL [t->dticks >= 0] @ /workdir/ncs/zephyr/kernel/timeout.c:108
[2020-05-18 23:51:56] 
[2020-05-18 23:51:56] ASSERTION FAIL [z_spin_lock_valid(l)] @ /workdir/ncs/zephyr/include/spinlock.h:44
[2020-05-18 23:51:56]   Recursive spinlock 0x20029574
[2020-05-18 23:51:56] [01:00:00.350,494] <err> os: r0/a1:  0x00000004  r1/a2:  0x0000002c  r2/a3:  0x00000008
[2020-05-18 23:51:56] [01:00:00.359,375] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000030 r14/lr:  0x00037603
[2020-05-18 23:51:56] [01:00:00.368,255] <err> os:  xpsr:  0x61000025
[2020-05-18 23:51:56] [01:00:00.373,626] <err> os: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]0
[2020-05-18 23:51:56] [01:00:00.384,307] <err> os: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]0
[2020-05-18 23:51:56] [01:00:00.394,989] <err> os: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]0
[2020-05-18 23:51:56] [01:00:00.405,670] <err> os: s[12]:  0x00000000  s[13]:  0xffffffff  s[14]:  0x00000000  s[15]0
[2020-05-18 23:51:56] [01:00:00.416,381] <err> os: fpscr:  0x224960a0
[2020-05-18 23:51:56] [01:00:00.421,722] <err> os: Faulting instruction address (r15/pc): 0x0003a1ce
[2020-05-18 23:51:56] [01:00:00.429,809] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[2020-05-18 23:51:56] [01:00:00.437,713] <err> os: Fault during interrupt handling
                                                                                  [2020-05-18 23:51:56] 
[2020-05-18 23:51:56] [01:00:00.444,274] <err> os: Current thread: 0x200294b4 (unknown)
[2020-05-18 23:51:56] [01:00:00.451,232] <err> cat_tracker: k_sys_fatal_error_handler, error: 4
[2020-05-18 23:51:56] [01:00:00.458,862] <err> cat_tracker: err_handler, error code: 4
[2020-05-18 23:51:56] ASSERTION FAIL [z_spin_lock_valid(l)] @ /workdir/ncs/zephyr/include/spinlock.h:44
[2020-05-18 23:51:56]   Recursive spinlock 0x2002956c
[2020-05-18 23:51:56] [01:00:00.475,830] <err> os: ***** HARD FAULT *****
[2020-05-18 23:51:56] [01:00:00.481,536] <err> os:   Fault escalation (see below)
[2020-05-18 23:51:56] [01:00:00.487,915] <err> os: r0/a1:  0x00000004  r1/a2:  0x0000002c  r2/a3:  0x00000008
[2020-05-18 23:51:56] [01:00:00.496,795] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000057 r14/lr:  0x000371cf
[2020-05-18 23:51:56] [01:00:00.505,706] <err> os:  xpsr:  0x6100000b
[2020-05-18 23:51:56] [01:00:00.511,047] <err> os: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]0
[2020-05-18 23:51:56] [01:00:00.521,728] <err> os: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]0
[2020-05-18 23:51:56] [01:00:00.532,379] <err> os: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]0
[2020-05-18 23:51:56] [01:00:00.543,060] <err> os: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]0
[2020-05-18 23:51:56] [01:00:00.553,741] <err> os: fpscr:  0x00000001
[2020-05-18 23:51:56] [01:00:00.559,082] <err> os: Faulting instruction address (r15/pc): 0x0003a1ce
[2020-05-18 23:51:56] [01:00:00.567,138] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[2020-05-18 23:51:56] [01:00:00.575,134] <err> os: Fault during interrupt handling
                                                                                  [2020-05-18 23:51:56] 
[2020-05-18 23:51:56] [01:00:00.581,726] <err> os: Current thread: 0x200294b4 (unknown)
[2020-05-18 23:51:56] [01:00:00.588,623] <err> cat_tracker: k_sys_fatal_error_handler, error: 0
[2020-05-18 23:51:56] [01:00:00.596,252] <err> cat_tracker: err_handler, error code: 0
[2020-05-18 23:51:56] ASSERTION FAIL [z_spin_lock_valid(l)] @ /workdir/ncs/zephyr/include/spinlock.h:44
[2020-05-18 23:51:56]   Recursive spinlock 0x2002956c
[2020-05-18 23:51:56] *** Booting Zephyr OS build v2.1.99-ncs1-2954-g479376885055  ***
[2020-05-18 23:51:56] Flash regions             Domain          Permissions
[2020-05-18 23:51:56] 00 02 0x00000 0x18000     Secure          rwxl

```

simensrostad, 2020-05-19:
Seems related: https://github.com/zephyrproject-rtos/zephyr/issues/13411

simensrostad, 2020-09-15:
This looks to be fixed now. @coderbyheart Could you confirm this?

coderbyheart, 2020-09-15:
Will have a look with the latest release.

coderbyheart, 2020-09-15:
Hard to test, because my log window gets swamped because of #69

coderbyheart, 2020-09-16:
With `mfw_nrf9160_1.2.1` and 2.5.1 the problem persists:

```
[00:00:14.279,785] <inf> cat_tracker: CLOUD_EVT_DATA_RECEIVED
Decoded message: {
        "version":      8011,
        "timestamp":    1600242800,
        "state":        {
                "cfg":  {
                        "act":  false,
                        "actwt":        299,
                        "mvt":  86400,
                        "acct": 5
                }
        },
        "metadata":     {
                "cfg":  {
                        "act":  {
                                "timestamp":    1600185174
                        },
                        "actwt":        {
                                "timestamp":    1600185174
                        },
                        "mvt":  {
                                "timestamp":    1600185174
                        },
                        "acct": {
                                "timestamp":    1600185174
                        }
                }
        }
}
[00:00:14.326,416] <inf> cloud_codec: SETTING ACTIVE TO: 0
[00:00:14.332,427] <inf> cloud_codec: SETTING ACTIVE WAIT TO: 299
[00:00:14.339,172] <inf> cloud_codec: SETTING MOVEMENT TIMEOUT TO: 86400
[00:00:14.346,435] <inf> cloud_codec: SETTING ACCEL THRESHOLD TIMEOUT TO: 5

...

[00:59:56.897,491] <inf> cat_tracker: Incoming RSRP status message, RSRP value is 37
[00:59:57.709,442] <inf> cat_tracker: Incoming RSRP status message, RSRP value is 32
[01:00:00.240,570] <dbg> date_time.date_time_handler: New date time update in: 3600 seconds
[01:00:00.249,511] <dbg> date_time.new_date_time_get: Updating date time UTC...
[01:00:00.257,385] <dbg> date_time.new_date_time_get: Time successfully obtained
[01:00:00.265,319] <inf> cat_tracker: DATE_TIME_OBTAINED_MODEM
[01:00:03.722,564] <inf> cat_tracker: Movement timeout triggered
[01:00:03.729,187] <inf> cat_tracker: The cat is moving!
[01:00:03.744,201] <dbg> nrf9160_gps.enable_gps: GPS mode is enabled
[01:00:03.762,420] <dbg> nrf9160_gps.gps_thread: A-GPS data update needed
[01:00:03.769,989] <inf> cat_tracker: GPS_EVT_AGPS_DATA_NEEDED
[01:00:03.762,908] <dbg> nrf9160_gps.start: GPS operational
[01:00:03.785,552] <dbg> nrf9160_gps.print_satellite_stats: Tracking: 0 Using: 0 Unhealthy: 0
```

closed 2020-09-16 by simensrostad

coderbyheart, 2020-09-16:
Fix confirmed.

commit d7497026 (2020-09-16): fix: Update movement timeout work upon change

Movement timeout handler is now re-schedueled if
there occurs a change in the movement timeout
parameter. Closes #46.