bifravst/firmware #69: GPS timeout configuration is not getting honored after change

coderbyheart, 2020-09-15:
Despite the `gpst` being set to `60` (seconds), GPS keeps trying to acquire a fix after the time has passed.

**Expected:** it should stop after 60 seconds.

```
[2020-09-15 17:53:43] [00:35:12.651,916] <dbg> aws_fota.on_publish_evt: received an unhandled MQTT publish event ona
[2020-09-15 17:53:43] [00:35:12.667,236] <inf> cat_tracker: CLOUD_EVT_DATA_RECEIVED
[2020-09-15 17:53:43] Decoded message: {
[2020-09-15 17:53:43]   "version":      8006,
[2020-09-15 17:53:43]   "timestamp":    1600185174,
[2020-09-15 17:53:43]   "state":        {
[2020-09-15 17:53:43]           "cfg":  {
[2020-09-15 17:53:43]                   "gpst": 60
[2020-09-15 17:53:43]           }
[2020-09-15 17:53:43]   },
[2020-09-15 17:53:43]   "metadata":     {
[2020-09-15 17:53:43]           "cfg":  {
[2020-09-15 17:53:43]                   "gpst": {
[2020-09-15 17:53:43]                           "timestamp":    1600185174
[2020-09-15 17:53:43]                   }
[2020-09-15 17:53:43]           }
[2020-09-15 17:53:43]   }
[2020-09-15 17:53:43] }
[2020-09-15 17:53:43] [00:35:12.693,328] <inf> cloud_codec: SETTING GPST TO: 60
[2020-09-15 17:53:43] Encoded message: {
[2020-09-15 17:53:43]   "state":        {
[2020-09-15 17:53:43]           "reported":     {
[2020-09-15 17:53:43]                   "cfg":  {
[2020-09-15 17:53:43]                           "gpst": 60
[2020-09-15 17:53:43]                   }
[2020-09-15 17:53:43]           }
[2020-09-15 17:53:43]   }
[2020-09-15 17:53:43] }
[2020-09-15 17:53:43] [00:35:12.836,456] <dbg> aws_fota.on_publish_evt: Received topic: $aws/things/352656100248049a
[2020-09-15 17:53:43] [00:35:12.847,473] <dbg> aws_fota.on_publish_evt: received an unhandled MQTT publish event ona
[2020-09-15 17:53:43] [00:35:12.862,426] <inf> cat_tracker: CLOUD_EVT_DATA_RECEIVED
[2020-09-15 17:53:43] Decoded message: {
[2020-09-15 17:53:43]   "version":      8006,
[2020-09-15 17:53:43]   "timestamp":    1600185174,
[2020-09-15 17:53:43]   "state":        {
[2020-09-15 17:53:43]           "cfg":  {
[2020-09-15 17:53:43]                   "gpst": 60
[2020-09-15 17:53:43]           }
[2020-09-15 17:53:43]   },
[2020-09-15 17:53:43]   "metadata":     {
[2020-09-15 17:53:43]           "cfg":  {
[2020-09-15 17:53:43]                   "gpst": {
[2020-09-15 17:53:43]                           "timestamp":    1600185174
[2020-09-15 17:53:43]                   }
[2020-09-15 17:53:43]           }
[2020-09-15 17:53:43]   }
[2020-09-15 17:53:43] }
[2020-09-15 17:53:43] [00:35:12.888,610] <inf> cat_tracker: No change in device configuration
[2020-09-15 17:53:44] [00:35:13.870,971] <inf> cat_tracker: Incoming RSRP status message, RSRP value is 38
[2020-09-15 17:53:49] [00:35:18.029,785] <dbg> nrf9160_gps.gps_thread: GPS has time window to operate
[2020-09-15 17:53:49] [00:35:18.037,628] <inf> cat_tracker: GPS_EVT_OPERATION_UNBLOCKED
[2020-09-15 17:53:49] [00:35:18.044,311] <dbg> nrf9160_gps.print_satellite_stats: Tracking: 0 Using: 0 Unhealthy: 0
[2020-09-15 17:53:49] [00:35:18.053,405] <dbg> nrf9160_gps.print_satellite_stats: Seconds since last fix 2118
[2020-09-15 17:53:50] [00:35:19.031,494] <dbg> nrf9160_gps.print_satellite_stats: Tracking: 0 Using: 0 Unhealthy: 0

[...]

[2020-09-15 17:57:09] [00:38:38.550,109] <dbg> nrf9160_gps.print_satellite_stats: Seconds since last fix 2318
```

simensrostad, 2020-09-15:
Are you using the latest fw version? I thought I adressed this issue.

coderbyheart, 2020-09-15:
Yes, that was using 2.5.1

simensrostad, 2020-09-16:
I meant modem firmware. I edited the first comment.

coderbyheart, 2020-09-16:
Right, `mfw_nrf9160_1.1.1`. Updating now.

coderbyheart, 2020-09-16:
With `mfw_nrf9160_1.2.1` and 2.5.1 the problem persists:

```
[00:00:14.279,785] <inf> cat_tracker: CLOUD_EVT_DATA_RECEIVED
Decoded message: {
        "version":      8011,
        "timestamp":    1600242800,
        "state":        {
                "cfg":  {
                        "act":  false,
                        "actwt":        299,
                        "mvt":  86400,
                        "acct": 5
                }
        },
        "metadata":     {
                "cfg":  {
                        "act":  {
                                "timestamp":    1600185174
                        },
                        "actwt":        {
                                "timestamp":    1600185174
                        },
                        "mvt":  {
                                "timestamp":    1600185174
                        },
                        "acct": {
                                "timestamp":    1600185174
                        }
                }
        }
}
[00:00:14.326,416] <inf> cloud_codec: SETTING ACTIVE TO: 0
[00:00:14.332,427] <inf> cloud_codec: SETTING ACTIVE WAIT TO: 299
[00:00:14.339,172] <inf> cloud_codec: SETTING MOVEMENT TIMEOUT TO: 86400
[00:00:14.346,435] <inf> cloud_codec: SETTING ACCEL THRESHOLD TIMEOUT TO: 5

...

[00:59:56.897,491] <inf> cat_tracker: Incoming RSRP status message, RSRP value is 37
[00:59:57.709,442] <inf> cat_tracker: Incoming RSRP status message, RSRP value is 32
[01:00:00.240,570] <dbg> date_time.date_time_handler: New date time update in: 3600 seconds
[01:00:00.249,511] <dbg> date_time.new_date_time_get: Updating date time UTC...
[01:00:00.257,385] <dbg> date_time.new_date_time_get: Time successfully obtained
[01:00:00.265,319] <inf> cat_tracker: DATE_TIME_OBTAINED_MODEM
[01:00:03.722,564] <inf> cat_tracker: Movement timeout triggered
[01:00:03.729,187] <inf> cat_tracker: The cat is moving!
[01:00:03.744,201] <dbg> nrf9160_gps.enable_gps: GPS mode is enabled
[01:00:03.762,420] <dbg> nrf9160_gps.gps_thread: A-GPS data update needed
[01:00:03.769,989] <inf> cat_tracker: GPS_EVT_AGPS_DATA_NEEDED
[01:00:03.762,908] <dbg> nrf9160_gps.start: GPS operational
[01:00:03.785,552] <dbg> nrf9160_gps.print_satellite_stats: Tracking: 0 Using: 0 Unhealthy: 0
```
```

simensrostad, 2020-09-16:
Is this comment meant for #46 ?

coderbyheart, 2020-09-16:
Right, sorry. This one is no longer valid with latest MFW.

simensrostad, 2020-09-16:
Closing this issue then.

closed 2020-09-16 by simensrostad