bifravst/firmware #78: FOTA does not download

coderbyheart, 2020-09-25:
```
[00:01:04.524,597] <inf> aws_fota: Checking for an available job
[00:01:04.532,226] <inf> aws_jobs: Subscribe: $aws/things/352656100248049/jobs/41fccd62-65dd-423b-9b6f-2fa1c98c7db0/update/#
[00:01:04.545,135] <inf> aws_fota: Subscribed to FOTA update topic $aws/things/352656100248049/jobs/41fccd62-65dd-423b-9b6f-2fa1c98c7db0/update/#
[00:01:04.764,831] <inf> aws_fota: Checking for an available job
[00:01:04.772,460] <inf> aws_jobs: Subscribe: $aws/things/352656100248049/jobs/41fccd62-65dd-423b-9b6f-2fa1c98c7db0/update/#
[00:01:04.785,369] <inf> aws_fota: Subscribed to FOTA update topic $aws/things/352656100248049/jobs/41fccd62-65dd-423b-9b6f-2fa1c98c7db0/update/#
[00:01:04.950,622] <inf> aws_jobs: Publish topic: $aws/things/352656100248049/jobs/41fccd62-65dd-423b-9b6f-2fa1c98c7db0/update
[00:01:04.962,646] <inf> aws_jobs: Publish payload {"status":"IN_PROGRESS","statusDetails": null,"expectedVersion": "1","clientToken": ""}
[00:02:03.589,996] <inf> cat_tracker: Device in PASSIVE mode
```

The job is marked as `IN_PROGRESS` but the download never starts.

coderbyheart, 2020-09-25:
If the device is moved then, it will no longer send updates:

```
[00:06:08.016,143] <inf> cat_tracker: The cat is moving!
[00:06:08.022,003] <inf> cat_tracker: Or it's lazy and this is just the movement timeout!

(no further output here)
```

coderbyheart, 2020-09-25:
The device keeps to operate normally.

coderbyheart, 2020-09-25:
```
[00:01:16.064,849] <dbg> aws_fota.on_publish_evt: Received topic: $aws/things/352656100248049/jobs/41fccd62-65dd-42d
[00:01:16.079,193] <inf> aws_fota: Start downloading firmware from bifravst-storagebucket1817f433-1jcq77cgkzd5h.s3.0
[00:01:16.233,764] <inf> download_client: Setting up TLS credentials, tag 42
[00:01:16.241,394] <inf> download_client: Connecting to bifravst-storagebucket1817f433-1jcq77cgkzd5h.s3.amazonaws.cm
[00:01:17.453,979] <err> download_client: Unable to connect, errno 95
[00:01:17.463,195] <err> aws_fota: Error (-95) when trying to start firmware download
[00:01:17.471,954] <err> aws_iot: aws_fota_mqtt_evt_handler, error: -95
[00:01:17.481,903] <inf> cat_tracker: CLOUD_EVT_DISCONNECTED
```

coderbyheart, 2020-09-25:
FW url: https://bifravst-storagebucket1817f433-1jcq77cgkzd5h.s3.amazonaws.com/41fccd62-65dd-423b-9b6f-2fa1c98c7db0

coderbyheart, 2020-09-25:
That is suspicious: 

> `<inf> download_client: Setting up TLS credentials, tag 42`

coderbyheart, 2020-09-25:
@sigvartmh might be able to help.

coderbyheart, 2020-09-26:
Actually this will then kill the cloud connection and the device will not be able to recover. This configuration does brick the ability to FOTA the device.

coderbyheart, 2020-09-26:
```
[2020-09-26 10:09:37] [00:00:08.310,943] <dbg> cat_tracker.device_config_send: No change in device configuration
[2020-09-26 10:09:37] [00:00:08.345,428] <dbg> aws_fota.update_job_execution: update_job_execution, state: 1, version_number: 5
[2020-09-26 10:09:37] [00:00:08.355,773] <inf> aws_jobs: Publish topic: $aws/things/352656100248049/jobs/41fccd62-65dd-423b-9b6f-2fa1c98c7db0/update
[2020-09-26 10:09:37] [00:00:08.367,797] <inf> aws_jobs: Publish payload {"status":"IN_PROGRESS","statusDetails": null,"expectedVersion": "5","clientToken": ""}
[2020-09-26 10:09:37] [00:00:08.665,710] <dbg> aws_fota.on_publish_evt: Received topic: $aws/things/352656100248049/jobs/41fccd62-65dd-423b-9b6f-2fa1c98c7db0/update/accepted
[2020-09-26 10:09:37] [00:00:08.680,084] <inf> aws_fota: Start downloading firmware from bifravst-storagebucket1817f433-1jcq77cgkzd5h.s3.amazonaws.com/41fccd62-65dd-423b-9b6f-2fa1c98c7db0
[2020-09-26 10:09:38] [00:00:09.804,870] <inf> download_client: Setting up TLS credentials, tag 42
[2020-09-26 10:09:38] [00:00:09.812,530] <inf> download_client: Connecting to bifravst-storagebucket1817f433-1jcq77cgkzd5h.s3.amazonaws.com
[2020-09-26 10:09:39] [00:00:10.404,327] <err> download_client: Unable to connect, errno 95
[2020-09-26 10:09:39] [00:00:10.413,360] <err> aws_fota: Error (-95) when trying to start firmware download
[2020-09-26 10:09:39] [00:00:10.422,058] <err> aws_iot: aws_fota_mqtt_evt_handler, error: -95
[2020-09-26 10:09:39] [00:00:10.432,556] <inf> cat_tracker: CLOUD_EVT_DISCONNECTED
[2020-09-26 10:09:39] [00:00:10.438,812] <err> aws_iot: publish_get_payload, error: -5
[2020-09-26 10:09:39] [00:00:10.945,770] <inf> cat_tracker: CLOUD_EVT_DISCONNECTED

[2020-09-26 10:10:30] [00:01:02.027,679] <dbg> nrf9160_gps.print_satellite_stats: Seconds since last fix 62
[2020-09-26 10:10:32] [00:01:04.036,376] <inf> cat_tracker: GPS_EVT_SEARCH_TIMEOUT
[2020-09-26 10:10:32] [00:01:04.042,633] <dbg> nrf9160_gps.stop_gps: Stopping GPS
[2020-09-26 10:10:32] [00:01:04.057,617] <inf> cat_tracker: GPS_EVT_SEARCH_STOPPED
[2020-09-26 10:10:32] [00:01:04.064,025] <inf> cat_tracker: Not connected to cloud!
[2020-09-26 10:10:32] [00:01:04.070,495] <inf> cat_tracker: Going to sleep for: 60 seconds
```

coderbyheart, 2020-09-26:
If this happens the device needs to be power cycled to recover (after removing all pending FOTA jobs). So this can bug can bricks devices in the field.

simensrostad, 2020-10-15:
This issue should now be resolved with https://github.com/nrfconnect/sdk-nrf/pull/3142.

sigvartmh, 2020-10-15:
I think you also need nrfconnect/sdk-nrf#3060 to ensure that you report failure to the AWS Job

simensrostad, 2020-10-19:
> I think you also need [nrfconnect/sdk-nrf#3060](https://github.com/nrfconnect/sdk-nrf/pull/3060) to ensure that you report failure to the AWS Job in case you actually end up trying to do a TLS download with no certificate stored so that you end up in the state above.

Ok, thanks!

coderbyheart, 2020-10-19:
I'll try it out this week.

commit 802a1f1b (2020-10-27): fix: update manifest

Adjust for upstream changes. Closes #78
closed 2020-10-27 by simensrostad