bifravst/firmware #81: Disconnect after JITP not handled

coderbyheart, 2020-10-07:
If the device is connecting for the first time, JITP will cause the connection to be terminated (see https://github.com/nrfconnect/sdk-nrf/pull/3017).

However this is also not handled in our application:

```
*** Booting Zephyr OS build v2.3.0-rc1-ncs1-3614-ga4ead9805140  ***
Flash regions           Domain          Permissions
00 02 0x00000 0x18000   Secure          rwxl
03 31 0x18000 0x100000  Non-Secure      rwxl

Non-secure callable region 0 placed in flash region 2 with size 32.

SRAM region             Domain          Permissions
00 07 0x00000 0x10000   Secure          rwxl
08 31 0x10000 0x40000   Non-Secure      rwxl

Peripheral              Domain          Status
00 NRF_P0               Non-Secure      OK
01 NRF_CLOCK            Non-Secure      OK
02 NRF_RTC0             Non-Secure      OK
03 NRF_RTC1             Non-Secure      OK
04 NRF_NVMC             Non-Secure      OK
05 NRF_UARTE1           Non-Secure      OK
06 NRF_UARTE2           Secure          SKIP
07 NRF_TWIM2            Non-Secure      OK
08 NRF_SPIM3            Non-Secure      OK
09 NRF_TIMER0           Non-Secure      OK
10 NRF_TIMER1           Non-Secure      OK
11 NRF_TIMER2           Non-Secure      OK
12 NRF_SAADC            Non-Secure      OK
13 NRF_PWM0             Non-Secure      OK
14 NRF_PWM1             Non-Secure      OK
15 NRF_PWM2             Non-Secure      OK
16 NRF_PWM3             Non-Secure      OK
17 NRF_WDT              Non-Secure      OK
18 NRF_IPC              Non-Secure      OK
19 NRF_VMC              Non-Secure      OK
20 NRF_FPU              Non-Secure      OK
21 NRF_EGU1             Non-Secure      OK
22 NRF_EGU2             Non-Secure      OK
23 NRF_DPPIC            Non-Secure      OK
24 NRF_GPIOTE1          Non-Secure      OK
25 NRF_REGULATORS       Non-Secure      OK

SPM: NS image at 0x18200
SPM: NS MSP at 0x200315e8
SPM: NS reset vector at 0x21e41
SPM: prepare to jump to Non-Secure image.
*** Booting Zephyr OS build v2.3.0-rc1-ncs1-3614-ga4ead9805140  ***
[00:00:00.200,042] <inf> watchdog: Watchdog timeout installed. Timeout: 60s
[00:00:00.207,580] <inf> watchdog: Watchdog started
[00:00:00.212,982] <inf> watchdog: Watchdog feed enabled. Timeout: 30000
[00:00:00.228,118] <inf> cat_tracker: ********************************************
[00:00:00.236,267] <inf> cat_tracker:  The cat tracker has started
[00:00:00.242,980] <inf> cat_tracker:  Version:     0.0.0-development-thingy91_nrf9160ns-ltem-debug
[00:00:00.252,593] <inf> cat_tracker:  Client ID:   352656100248049
[00:00:00.259,399] <inf> cat_tracker:  Endpoint:    a34x44yyrk96tg-ats.iot.eu-central-1.amazonaws.com
[00:00:00.269,195] <inf> cat_tracker: ********************************************
[00:00:00.277,343] <dbg> nrf9160_gps.open_socket: GPS socket created, fd: 1232491587
[00:00:00.285,705] <inf> cat_tracker: Configuring the modem...
[00:00:01.668,609] <dbg> cat_tracker.modem_rsrp_handler: Incoming RSRP status message, RSRP value is 38
[00:00:01.678,833] <dbg> cat_tracker.modem_rsrp_handler: Incoming RSRP status message, RSRP value is 39
[00:00:01.689,361] <dbg> cat_tracker.lte_evt_handler: LTE cell changed: Cell ID: 35493142, Tracking area: 2305
[00:00:01.700,134] <dbg> cat_tracker.lte_evt_handler: PSM parameter update: TAU: -1, Active time: -1
[00:00:01.755,981] <dbg> cat_tracker.lte_evt_handler: RRC mode: Connected
[00:00:02.861,389] <inf> cat_tracker: Network registration status: Connected to roaming network
[00:00:02.870,727] <dbg> date_time.new_date_time_get: Updating date time UTC...
[00:00:02.878,601] <dbg> date_time.current_time_check: Date time never set
[00:00:02.886,016] <dbg> date_time.new_date_time_get: Current time not valid
[00:00:02.893,615] <dbg> date_time.new_date_time_get: Fallback on cellular network time
[... 114 lines cut ...]
[00:00:56.074,829] <dbg> nrf9160_gps.print_satellite_stats: Seconds since last fix 56
[00:00:57.065,643] <dbg> nrf9160_gps.print_satellite_stats: Tracking: 0 Using: 0 Unhealthy: 0
[00:00:57.074,859] <dbg> nrf9160_gps.print_satellite_stats: Seconds since last fix 57
[00:00:58.065,429] <dbg> nrf9160_gps.print_satellite_stats: Tracking: 0 Using: 0 Unhealthy: 0
[00:00:58.074,554] <dbg> nrf9160_gps.print_satellite_stats: Seconds since last fix 58
[00:00:59.065,368] <dbg> nrf9160_gps.print_satellite_stats: Tracking: 0 Using: 0 Unhealthy: 0
[00:00:59.074,493] <dbg> nrf9160_gps.print_satellite_stats: Seconds since last fix 59
[00:01:00.065,765] <dbg> nrf9160_gps.print_satellite_stats: Tracking: 0 Using: 0 Unhealthy: 0
[00:01:00.074,981] <dbg> nrf9160_gps.print_satellite_stats: Seconds since last fix 60
[00:01:00.220,306] <dbg> watchdog.primary_feed_worker: Feeding watchdog
[00:01:01.065,429] <dbg> nrf9160_gps.print_satellite_stats: Tracking: 0 Using: 0 Unhealthy: 0
[00:01:01.074,554] <dbg> nrf9160_gps.print_satellite_stats: Seconds since last fix 61
[00:01:02.065,460] <dbg> nrf9160_gps.print_satellite_stats: Tracking: 0 Using: 0 Unhealthy: 0
[00:01:02.074,584] <dbg> nrf9160_gps.print_satellite_stats: Seconds since last fix 62
[00:01:04.083,251] <inf> cat_tracker: GPS_EVT_SEARCH_TIMEOUT
[00:01:04.089,599] <dbg> nrf9160_gps.stop_gps: Stopping GPS
[00:01:04.104,553] <inf> cat_tracker: GPS_EVT_SEARCH_STOPPED
[00:01:04.110,961] <inf> cat_tracker: Not connected to cloud!
[00:01:04.117,431] <inf> cat_tracker: Going to sleep for: 60 seconds
[00:01:04.234,741] <inf> cat_tracker: Board is running expected modem firmware version: mfw_nrf9160_1.2.1
[00:01:04.244,934] <dbg> cat_tracker.modem_buffer_populate: Entry: 1 of 19 in modem buffer filled
[00:01:04.254,394] <dbg> cat_tracker.battery_buffer_populate: Entry: 1 of 19 in battery buffer filled
[00:01:04.265,655] <dbg> cat_tracker.sensors_buffer_populate: Entry: 1 of 19 in sensor buffer filled

```

simensrostad, 2020-10-19:
This will be handled by the reconnection routine in the new firmware architecture.

coderbyheart, 2020-10-27:
This blocks #3 because the device will never connect to the cloud, since all tests use a fresh set of credentials

coderbyheart, 2020-11-16:
Fixed in a90d0057d6046a1d0683f8f935f22ab1cf9820e0

closed 2020-11-16 by coderbyheart