bifravst/firmware #136: Problems with gitbook.io

altishchenko, 2020-11-09:
Hi Markus!
Looks like something happened to bifravst.gitbook.io - 404 on every link :(

coderbyheart, 2020-11-09:
Yes, this is due to bifravst/bifravst#192.

You can always find the latest documentation at https://bifravst.github.io/ (this is also the link on top of https://github.com/bifravst).

closed 2020-11-09 by coderbyheart

coderbyheart, 2020-11-10:
P.S.: I understand that it can be annoying to have older links now working, but this project is still in a pre-release phase, so I believe it is OK to take some liberties. Once bifravst/bifravst/issues/56 is completed the docs will have a permanent home.

In any case, I will always make sure the https://bifravst.github.io/ always redirects to the latest docs.

altishchenko, 2020-11-10:
Nah, no problem :)